pimoroni/hyperpixel #1: Loading kernel module fails at boot

shavs, 2017-06-10:
# Issue
On RetroPie 4.2 (latest as of writing), loading the kernel module (second line of the [/boot/config.txt file here](https://github.com/pimoroni/hyperpixel#lcd-display)) fails if the distribution is not Raspbian. Whilst this isn't *the* most important issue, it would be nice to be able to use this display on other distributions

# Temporary Solution until Resolved
Creating a new file at /etc/init.d/, such as /etc/init.d/hyperpixel-init.sh, and putting:
```bash
#!/bin/bash
./usr/bin/hyperpixel-init
```
Obviously, there are error and startup checks that could be implemented here, but this is the quickest fix I came across this morning after receiving the display. 
I have not tested touchscreen support for the display on other distributions as of yet.

jasarien, 2017-06-10:
I'm getting this exact same error on the version of Raspbian (with PIXEL) freshly installed via NOOBS.

Gadgetoid, 2017-06-10:
It looks like my `initramfs` solution for initialising the touchscreen as early as possible is perhaps too fragile to rely upon. D'oh!

With the help of @ali1234 I have put together some systemd init scripts to replace both the Touchscreen and LCD init processes. They're currently available in testing and are installed by `setup.sh` like so: https://github.com/pimoroni/hyperpixel/blob/testing/setup.sh#L216-L218

I plan to refactor the `hyperpixel-touch` script into a proper, daemonized Python script in the same pattern as our Picadehatd driver. It wont make any meaningful difference to end users, it's just more correct :D

johnmccabe, 2017-06-14:
@Gadgetoid I was seeing something similar with the `hyperpixel-touch` bombing out when it hit:
```
bus = smbus.SMBus(3)
```
This was on a completely fresh install, with the first action on the node being to run the single line installer.
I noticed that `/dev/i2c-3` was missing which would explain the bus initialisation step above failing so I ran `raspi-config` enabled `i2c` then immediately disabled it. Running `hyperpixel-touch` now worked and I was able to use the display as expected.

RogueM, 2017-06-14:
the systemd implementation is now available in the master branch.

closed 2017-06-14 by RogueM