pimoroni/hyperpixel #6: Absence of /dev/i2c-3 causes hyperpixel-touch to fail

johnmccabe, 2017-06-14:
On a completely fresh Raspbian install, with the first action on the node being to run the single line installer. I was observing `hyperpixel-touch` bombing out when it hit:
```
bus = smbus.SMBus(3)
```
I noticed that `/dev/i2c-3` was missing which would explain the bus initialisation step above failing so I ran `raspi-config` enabled `i2c` then immediately disabled it. Running `hyperpixel-touch` now worked and I was able to use the display as expected.

RogueM, 2017-06-14:
are you on Raspbian Lite? if so, this is expected... the touch device is not relevant to that environment.

RogueM, 2017-06-14:
this is fixed now, assuming you are using Raspbian Lite... but obviously this does not change the fact that the touch capability of the screen is no use in that environment.

Still, the boot messages should be free from errors related to it now. Thanks for reporting the issue!

closed 2017-06-14 by RogueM

johnmccabe, 2017-06-14:
@RogueM I've just this minute set it up on Raspbian Lite and the framebuffer seems to work well, I've not tried touch (yet) so have disabled the service.

One thing I did observe however was that on first run of the setup it threw an error running modprobe (I didn't catch the name of the module it complained about, can try again later), running it a second time completed without error and the display worked on reboot.