pimoroni/hyperpixel #7: Install script assumes Python is already installed

ngreatorex, 2017-06-17:
I don't use the standard Raspbian full image, but instead a custom debootstrapped minimal image. As such, I don't already have Python installed. The install script should either fail if python2.7 isn't installed or instead automatically install it.

Gadgetoid, 2017-06-19:
Install scripts are ostensibly for only Raspbian - although they do support Raspbian-based distributions and some other OSes. We'll look into this, but can't support every corner case. It's assumed if you know what you're doing enough to bootstrap a system without Python then you're probably comfortable picking and choosing the components you want from the manual setup.

ngreatorex, 2017-06-19:
That's a fair point, but I still think there should be a guard to stop it leaving your system in a broken state. I will look at putting together a PR this evening to fix it.

RogueM, 2017-06-19:
I am fixing this... however, the setup already installs pythin packages that depends on a minimal python system, sp if this fails currently then I think the issue is more complex than described.

ngreatorex, 2017-06-19:
@RogueM 

The issue was that the "dpkg -i" call for python-evdev leaves the system in a broken state (as dpkg doesn't handle dependencies), so all the subsequent "apt-get install" calls fail to install anything and just complain. 

The python-evdev deb you provide has "Depends: libc6 (>= 2.4), python (>= 2.7), python (<< 2.8)". As it depends on python and not python-minimal, the change you committed won't quite fix it. The python package is simply there for dependencies, so installing python rather than python-minimal uses only a few bytes more (it's not python-all).

RogueM, 2017-06-20:
ah, Ok... I blame @phil who made the edev oackage a while back... I doubt that it needs more than python-minimal but we'll look into. Ultimately there is no pont trying to support a system where pyrthon is absent but then install everything but the kitchecn sink ;-)

ngreatorex, 2017-06-20:
@RogueM the python package is the correct dependency. It's a more or less empty package that itself depends on python-minimal. The "'kitchen sink" package, python-all, is separate.

RogueM, 2017-06-22:
well, like I said initially, your problem is more complex than described. The `python` package is in the recommends list for `python-minimal` so it should really be pulled in automatically. But I can change that, and make it explicit, sure.

closed 2017-06-22 by RogueM