pimoroni/hyperpixel #8: Dual screen setup question

aCorrado, 2017-06-26:
Hi,
I was thinking to have a setup like:
- a raspicam
- hyperpixel to be used as a secondary display (always on, using a framebuffer to push data to it)
- a tv attached to HDMI as the main screen

Do you think this is possible?
Thanks

Gadgetoid, 2017-06-26:
I think your quest is related to: http://forums.pimoroni.com/t/hdmi-and-hyperpixel/5122 

To which I don't have an answer, yet!

The forum poster claims the VGA666 can be used as a secondary screen. I wasn't aware it was possible to productively use a DSI/DPI display alongside HDMI, since in both cases they take over `/dev/fb0` and are driven directly from the GPU- which we can't affect or change since the firmware is a closed-source black box. 

I'm still pretty sure it's impossible to use HDMI + HyperPixel or HDMI + VGA666 or HDMI + Official Touchscreen in any meaningful way.  But I'd welcome any wisdom to prove me wrong!

Sources: https://www.raspberrypi.org/forums/viewtopic.php?t=120541
https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=94424&sid=0ea9214634fc02b0f0dd1fcafa940296&start=125

Quote from Gordon back in 2015:

> I believe all you need to do first is to create a second fbdev driver. This can then be used for the second X windows display (I'm guessing a lot here).

> The firmware sets up the HDMI and LCD displays and the primary display gets a frame buffer set up. The details of this buffer is read from the GPU through the mailbox interface when the fbdev is being initialised, check out the device-tree contents to see which drivers access the firmware driver. So what we really need is a second fbdev to load up the second frame buffer from the firmware...

> This will require that we add the second buffer information to the firmware, but that sounds quite easy in comparison to everything else!

> Once a second fbdev has been created I _think_ X windows will use it automatically (or it may require extra setup)

> Gordon

As far as I know, nobody has done this yet.

Cynar, 2017-07-06:
I know the official screen can work alongside an hdmi display, to an extent. Id can't be used as part of the desktop, but you can route OMXplayer to the hdmi.

I am using this setup on a test project. The touchscreen shows a control panel, while the hdmi is used for a video feed.

dglaude, 2018-03-17:
For me, the following `omxplayer -p -o hdmi big.avi` does not output anything on the HDMI side, but display on the Hyperpixel.

Any information on how is that done on your project or somebody succeeding this on the HyperPixel is welcome.

Gadgetoid, 2019-08-29:
Pi 4 + Buster fixes this in probably the most comprehensive way we're ever going to see it fixed. I'd wager due to the change of GPU we're unlikely to see this backported to earlier Pi models.

closed 2019-08-29 by Gadgetoid

jvelo, 2021-01-24:
Sorry in advance for reviving an old thread ; are there any pointers on how one would have a go at this with Buster and a Pi4 ?

elixx, 2021-09-06:
I am also trying to get a HyperPixel display to work on a RPI400 in addition to the normal HDMI out. It doesn't have to be an X display, but just looking to output / draw to it.