pimoroni/hyperpixel #9: Using Hyperpixel with LED output / Button input

mikeparsons84, 2017-07-30:
Looking at the pinout the hyperpixel understandably uses all the pins so there are no free pins to run a button or LED. I've looked at a GPIO port expander, which also would have the same issues.

So I'm thinking is there a way to directly connect the hyperpixel RPi to a second RPi directly over Ethernet to a second RPi just to control its GPIO and connect the buttons / LEDs to the second one.

Any thoughts on how I can get around this?

Gadgetoid, 2017-07-31:
Hyper Pixel uses two of the available pins as a software i2c bus (which shows up as /dev/i2c-2 iirc). If you can tap into these pins then you can connect pretty much any i2c sensor, expander or device you like.

See the pins marked "Touch Data" and "Touch Clock" here: https://pinout.xyz/pinout/hyperpixel

dandrews7396, 2017-08-01:
@mikeparsons84 I would encourage you to check out the last version of the gpiozero library by @bennuttall. Read more about it here: http://bennuttall.com/whats-new-gpio-zero-v1-4/

bennuttall, 2017-08-01:
Definitely! GPIO Zero supports remote pins, so you can execute code which uses the hyperpixel directly, and controls devices connected to another Pi on the network.

From the docs:
- [configuring remote gpio](http://gpiozero.readthedocs.io/en/stable/remote_gpio.html)
- [remote gpio examples](http://gpiozero.readthedocs.io/en/stable/recipes_remote_gpio.html)
- [pigpio pin reference](http://gpiozero.readthedocs.io/en/stable/api_pins.html#module-gpiozero.pins.pigpio)

lidense, 2020-11-30:
Hi, did you succeed in using gpios on an rpi running hyperpixel?
I like that screen because it has good contrast but would like to use a UPS at the same time to have safe shutdown etc. any suggestions?
Thanks
Dave

closed 2023-01-20 by Gadgetoid