pimoroni/hyperpixel #10: Feature request

PalimPalimPalim, 2017-08-01:
Add some example code on how to add 2 to 3 physicals buttons via i2c.

Gadgetoid, 2017-08-01:
If size isn't a constraint, the easier thing to do is use a Picade PCB.

Or something like: https://shop.pimoroni.com/products/adafruit-ft232h-breakout-general-purpose-usb-to-gpio-spi-i2c

Or any generic i2c IO expander. A common and well-documented hobbyist part is the MCP23017.

And USB-compatible Arduino (Leonardo-style) or micro dev board would also work, via USB serial or HID joystick/keyboard emulation, to add buttons.

I'll see what I can get written up regarding these solutions, or perhaps beg @sandyjmacdonald for another epic tutorial ;)

Bauingenieur, 2017-08-22:
Hi I'm new here!
As i read it right, there is no possibility to plug in an expander or replicator for using HyperPixel and an DAC HAT?

steffenmllr, 2017-09-04:
I was just running into the same Problem. I couldn't find a i2c IO Expander on pimoroni so I bought this one: https://www.abelectronics.co.uk/p/71/IO-Pi-Zero 

Any help setting it up with my hyperpixel and raspberry zero would be appreciated - seems like a lot of people are interested in that topic

eighthree, 2017-11-19:
I attached usb hub to use with my hyperpixel and went the micro dev board route via USB serial.

I have written an example arduino sketch that uses a feather m0 express as a USB HID device for my retropie build, which also does power monitoring (enables a battery indicator on a retropie build).

https://gist.github.com/eighthree/408075d4888b1bd3cc4652b2ed952e78
https://gist.github.com/eighthree/cc8c640cb6b0898cfc6d7130f8ae3f4d

ismangil, 2017-12-16:
I found this as well https://shop.pimoroni.com/products/adafruit-atsamd09-breakout-with-seesaw to get back expansion capabilities with HyperPixel, to be connected to the Touch I2C bus 3. 

https://learn.adafruit.com/adafruit-seesaw-atsamd09-breakout?view=all

ferazambuja, 2018-05-31:
The way a pi zero is affordable it might easy to use it as GPIO expander:
https://www.raspberrypi.org/magpi/pi-zero-gpio-expander/

If anyone has tried let me know.

closed 2019-08-27 by Gadgetoid