pimoroni/hyperpixel #11: Feature request: controllable backlight intensity

TheDiveO, 2017-08-05:
After reading through the product page and the sources here multiple times I'm still not sure if the HyperPixel hardware supports controlling the backlight intensity, albeit the product page somehow hints at this.

Anyway, if the HW is capable, please allow control of the backlight intensity.

At the moment, the backlight in `/sys/class/backlight/rpi_backlight/brightness` can only be set to `0`(off) and `1` (on), as also indicated by `/sys/class/backlight/rpi_backlight/max_brightness`.

Gadgetoid, 2017-08-07:
There is actually a PWM backlight module that I'm in the process of asking for inclusion in Raspbian to make this possible. At the moment getting the PWM backlight up and running requires manual compilation of `pwm_bl.ko`, which (from experience with RP USB Display) is not a good long-term solution.

See here for progress: https://github.com/raspberrypi/linux/issues/2063

koehnens, 2018-03-22:
Are there any solutions at the moment to control the backlight intensity?
Even if it requires manual work, it's better than not being able to use it at all.

bfanger, 2018-11-10:
**A workaround:**

Both the [hyperpixel](https://pinout.xyz/pinout/hyperpixel) and the [hyperpixel4](https://pinout.xyz/pinout/hyperpixel4) use pin  19 (BCM).

Install wiringpi if it's not yet installed. _(sudo apt install wiringpi)_

```bash
# Set the pin to PWM mode
gpio -g mode 19 pwm 
# Set the value/brightness to 60
gpio -g pwm 19 60
```

| Value | Effect |
| ------- | ---|
| 0 - 24 | Backlight off |
| 25 - 90 | Various degrees of brightness |
| 91 - 1023 | Full brightness |

fireslayer26, 2019-03-14:
What do you do after installing wiringpi? I am using the Hyperpixel4 screen in my gaming handheld. Can I control the brightness through hotkey inputs, like "Select + Left or right"? Thanks

closed 2019-08-27 by Gadgetoid

herndonj, 2019-12-21:
> **A workaround:**
> 
> Both the [hyperpixel](https://pinout.xyz/pinout/hyperpixel) and the [hyperpixel4](https://pinout.xyz/pinout/hyperpixel4) use pin 19 (BCM).
> 
> Install wiringpi if it's not yet installed. _(sudo apt install wiringpi)_
> 
> ```shell
> # Set the pin to PWM mode
> gpio -g mode 19 pwm 
> # Set the value/brightness to 60
> gpio -g pwm 19 60
> ```
> 
> Value	Effect
> 0 - 24	Backlight off
> 25 - 90	Various degrees of brightness
> 91 - 1023	Full brightness

I have wiringpi installed and all seems to be fine on running these commands, but the brightness does not change.  Has anyone had any luck with this?  All suggestions welcome on whee this code (or other code) could work.

tinspin, 2020-02-08:
You need to use the latest wiringpi:

wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb

Ryce358, 2020-04-03:
@tinspin (or anyone) I have used the following in a python script so I can dynamically control the backlight on my HyperPixel

`import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BOARD)
GPIO.setup(19, GPIO.OUT)
pwm = GPIO.PWM(19, 60)

pwm.start(100) # brightness 0-100`

As you state, it breaks the auto-backlight feature.  Any idea how I can reverse this and get back to auto-backlight?

Also, after ending my python session in terminal and starting a new one, that same script does nothing.  I'm currently stuck with a blank screen! Oops!

JakeCover, 2020-07-02:
> @tinspin (or anyone) I have used the following in a python script so I can dynamically control the backlight on my HyperPixel
> 
> ```
> import RPi.GPIO as GPIO
> 
> GPIO.setmode(GPIO.BOARD)
> GPIO.setup(19, GPIO.OUT)
> pwm = GPIO.PWM(19, 60)
> 
> pwm.start(100) # brightness 0-100
> ```
> 
> As you state, it breaks the auto-backlight feature. Any idea how I can reverse this and get back to auto-backlight?
> 
> Also, after ending my python session in terminal and starting a new one, that same script does nothing. I'm currently stuck with a blank screen! Oops!
> 
> Edit: after rebooting the backlight is back to auto but it would be good to be able to do this without rebooting the pi. Thanks

I've found that the `RPi.GPIO` library is pretty bad for this, at least on a Pi 0 W. For me, it created a bunch of flickering issues and just looked really bad. I don't fully understand this, but it seems that `RPi.GPIO` is using software PWM, which is extremely limited because of linux process management or something. Instead. you should be using hardware PWM, which is a ton more stable. I guess it uses some other mechanic, maybe another chip on the board? IDK, but it works so much better. I'm using `pigpio`, found here: http://abyz.me.uk/rpi/pigpio On the installation tab there are setup instructions. I downloaded it from GitHub and used the makefile, I can't vouch for the apt package. It comes with a python library, which basically just works, and really well. Here's my code:
```python
import pigpio
gpio = pigpio.pi()
gpio.set_PWM_dutycycle(19, 64)  #0-255, so 64 is 1/4 duty cycle
```
Be sure to run `sudo pigpiod` to start the daemon before you run this script. Good luck!

Ryce358, 2020-07-07:
Thanks @JakeCover - At first it didn't work but a trusty reboot and now I'm flicker free!

carcamerarig, 2020-09-28:
Hi there, is this actually working now, flicker free and able to use the autobacklight feature?
Thanks
J

walletje-w, 2021-04-05:
@carcamerarig as far as I know there is no light sensor build into this device so autobacklight will never work on its own. 

I've decided to link a VCNL4010 over i2c to adjust the brightness based on the amount of light in the room. Then in combination with the code from @JakeCover it works flicker free and the build in 'autobacklight' still works. Which means it goes to black after a certain amount of no activity and when you tap it, it wakes up. 

So my order of how I did things

1. First installing the Hyperpixel: https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-hyperpixel-4
2. Enabling i2c - In my case it wasn't `sudo ln -s /dev/i2c-3 /dev/i2c-1` but `sudo ln -s /dev/i2c-11 /dev/i2c-1`
3. Install Blinka https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi
4. Then the VCNL4010 library https://learn.adafruit.com/using-vcnl4010-proximity-sensor/python-circuitpython
5. To test it I used the adafruit code in combination with the code from Jakecover above and placed it in a document testvcnl4010.py (For ease of use I've placed the code at the bottom of this post)
6. `sudo pigpiod`
7. `python3 testvcnl4010.py`

So in this case I've put a simple if else statement to test if this works. After this I will make it more gradual that it slowly steps down or up depending on the amount of Lux in the room. 

- 30 is really dim, perfect for when you get out of bed while it is still dark
- 128 is pretty bright
- 255 is standard brightness

I did notice that when you dimm the screen it starts to make more noise (a high pitched sound). It always makes it, but just a tad more. 

```
# SPDX-FileCopyrightText: 2017 Tony DiCola for Adafruit Industries
# SPDX-License-Identifier: MIT

# Simple demo of the VCNL4010 proximity and light sensor.
# Will print the proximity and ambient light every second.
import time

import board
import busio

import adafruit_vcnl4010

# library for the brightness adjustment based on the VNCL4010 sensor
import pigpio
gpio = pigpio.pi()

# Initialize I2C bus and VCNL4010 module.
i2c = busio.I2C(board.SCL, board.SDA)
sensor = adafruit_vcnl4010.VCNL4010(i2c)

# You can optionally adjust the sensor LED current.  The default is 200mA
# which is the maximum value.  Note this is only set in 10mA increments.
# sensor.led_current_mA = 120  # Set 120 mA LED current

# You can also adjust the measurement frequency for the sensor.  The default
# is 390.625 khz, but these values are possible to set too:
# - FREQUENCY_3M125: 3.125 Mhz
# - FREQUENCY_1M5625: 1.5625 Mhz
# - FREQUENCY_781K25: 781.25 Khz
# - FREQUENCY_390K625: 390.625 Khz (default)
# sensor.frequency = adafruit_vcnl4010.FREQUENCY_3M125  # 3.125 Mhz

# Main loop runs forever printing the proximity and light level.
while True:
        proximity = sensor.proximity  # Proximity has no units and is a 16-bit
    # value.  The LOWER the value the further
    # an object from the sensor (up to ~200mm).
        print("Proximity: {0}".format(proximity))
        ambient_lux = sensor.ambient_lux
        print("Ambient light: {0} lux".format(ambient_lux))
        if ambient_lux < 100:
                print("really dark")
                gpio.set_PWM_dutycycle(19, 30)  #0-255, so 64 is 1/4 duty cycle
        elif ambient_lux > 100 and ambient_lux <300:
                print("evening")
                gpio.set_PWM_dutycycle(19, 128)  #0-255, so 64 is 1/4 duty cycle
        else:
                print("daylight")
                gpio.set_PWM_dutycycle(19, 255)  #0-255, so 64 is 1/4 duty cycle
        time.sleep(1.0)
```

CatalinMustata, 2021-05-09:
@walletje-w thanks a lot for your work man! Got me going on this and I managed to fix the whining by increasing the PWM frequency to 20khz (just outside of the audible range), using the hardware PWM from `pigpio` . Duty cycle is a bit weird in that mode, but as far as I can see, it reacts properly to values between 30000 (anything lower will turn off the display or flicker) and 100000. I might be wrong but it seems to prefer round values that are multiple of 10k.

I'll try my hand at doing gradual backlight adjustments (from current value to target value) soon.

walletje-w, 2021-05-17:
Great that you've managed to fix the whining! At the moment my setup is in use for a different experiment but when I will continue I'll try to implement your solution. Do you have any more info on how you changed the PWM?

CatalinMustata, 2021-05-22:
Sure thing. Here's my repo: https://github.com/CatalinMustata/backlight-control - it's NodeJS, but uses same `pigpio` lib, so you can probably find a Python equivalent quickly.

Main idea is to use `hardwarePwmWrite(freq, dutyCycle)`. I'm using `20000` for the frequency now, but I want to play around with it when I have some time. Maybe lower it drastically, to something like 1000hz where even if it whines, it shouldn't have enough energy to become audible. End goal is to get the display dimmer, if possible (lowest setting before it turns off still seems too bright)

ayufan, 2021-06-18:
The best is to modify `dts` to expose `pwm-backlight`. This is then available via `/sys/class/backlight` and can properly be supported by any desktop env without hacks.

Gadgetoid, 2021-06-18:
> The best is to modify `dts` to expose `pwm-backlight`. This is then available via `/sys/class/backlight` and can properly be supported by any desktop env without hacks.

I never actually had any success doing this, due to having to feed pwm-backlight a clock or somesuch. I don't think the module is shipped with Raspbian, I got about this far petitioning for its inclusion - https://github.com/raspberrypi/linux/issues/2063

CatalinMustata, 2021-06-23:
I'd give that a whirl if/when I'll have some time, just out of curiosity, even if I don't think it would work in my use case (control backlight from VCNL4010 light sensor).

@Gadgetoid: out of curiosity (and need), is there any way of dimming the display further than 28-ish %? It feels like it could use a bigger capacitor on the backlight power rail or something.

Gadgetoid, 2021-06-30:
One option is to remove the dtoverlay for handling the backlight, and use a PWM-supporting library or service for the Pi instead.

F1p, 2021-11-06:
Any further developments? 

Looking at trying to use xset blanking to trigger display backlight off

bolsoncerrado, 2022-11-17:
Hi,

I do own several HP4 and one of them is truly too bright almost burning whiteish contents....

1) how can I determine which level of brightness any other of my HP4 are set to?
and of course:
2) how do I match the whiteish HP4 to the others? Permanent fix would be great. I don't need room light conditions nor change brightness dinamically nor anything other than one-time set correct brightness.

THanks!

MattGrayYes, 2023-01-09:
> **A workaround:**
>
> Both the [hyperpixel](https://pinout.xyz/pinout/hyperpixel) and the [hyperpixel4](https://pinout.xyz/pinout/hyperpixel4) use pin 19 (BCM).
>
> Install wiringpi if it's not yet [...]

This works great for me thanks!
I've written it into this shell script for easy use: [brightness.sh](https://gist.github.com/MattGrayYes/eb753ae20b611b3c57aeaf933f128aeb)

bolsoncerrado, 2023-01-09:
Thank you Sir!

ayufan, 2023-01-09:
@bolsoncerrado `bash`.

bolsoncerrado, 2023-01-09:
Ty bud!

Torxed, 2023-09-04:
`wiringpi` is dead/deprecated.
Instead, [pigpio](http://abyz.me.uk/rpi/pigpio/) is recommended.

To use it, you need to:
1. Start the daemon with `# pigpiod`
2. Set [PWM duty cycle](http://abyz.me.uk/rpi/pigpio/pigs.html#HP) on pin `19` with `$ pigs hp 19 100 600000`