pimoroni/hyperpixel #15: Feature: Driver for Android Things

Malo87, 2017-09-07:
I'm quite sure that this display can sell a lot if it's usable with a user space driver for Android Things
https://github.com/androidthings/contrib-drivers

Gadgetoid, 2017-09-13:
This is, unfortunately, not something we've got the time or expertise to pull off ourselves.

However, I have, if I remember correctly, mentioned it in passing to @proppy. I believe it may be worthwhile raising an issue asking for HyperPixel support.

closed 2017-09-13 by Gadgetoid

Malo87, 2017-09-13:
Thanks for the reply

proppy, 2017-09-14:
@Malo87 currently there is no "user space driver" model for creating new display but that's something that would be interesting to raise on http://b.android.com/.

Do we have more details on what the `hyperpixel-init` binary does, that would help to understand if that can easily be supported.