pimoroni/hyperpixel #16: Raspberry Pi zero W display just flicker

VladoPortos, 2017-10-10:
Hello,
I just installed 2017-09-07-raspbian-stretch-lite on rPi Zero W + hyperpixel, run the script and after reboot the screen is semi working, just flickers displaying the console and then disappearing its very strange... I have tried 2 displays now and both doing the same... I'm using original USB power adapter from rPi 

Any idea what it can be ?

VladoPortos, 2017-10-10:
Ok seems to be issue with one of my rpi zero w, tested on another one and works ok !

VladoPortos, 2017-10-10:
literally 5 min later the display start flickering again...

VladoPortos, 2017-10-10:
Well after some more testing I'm worried to try with another rPi W... I did try 3 of them now. What I have noticed is that it starts ok, holds the display fine... and than start flickering, after that happens the rPi flickers  on any hyperpixel display right away ( I have 5 of each )... which to me suggest something fried :-/ nobody experience the same ? Was it tested properly with the zero line of rpi ?

VladoPortos, 2017-10-10:
Well after some more testing... this is not a good news. Display is killing something in rPi zero w, every new rPi zero w I plug in will work like for 3 min top and then the display stop working. Mind you the display is ok, if plugged to another rPi zero w it will work for a while until it dies again. Once this happens the rpi works still, it will boot up, but it will not work with hyperpixel again... somebody owns me 4x rPi zero w 👎  if they won't work with 480x Chinese TFT also :( I'm so disappointed with this :(

Gadgetoid, 2017-10-11:
This is definitely not expected behaviour!

I believe you're currently in contact with Matt about this? Hopefully he'll have you sorted quickly.

closed 2017-10-11 by Gadgetoid