pimoroni/hyperpixel #18: Larger console font

snobu, 2017-11-22:
Is there a way to embiggen the (framebuffer) console font?

/etc/defaults/console-setup:

```
FONTFACE="Terminus"
FONTSIZE="32x16"
```

32 is max supported by Linux for the console.. so i guess i need to lower the HyperPixel resolution. How would i do that? Can i scale to something like 640x200, is there a scaler built-in the display?

I just need very large fonts for a demo, i would never ruin the super high res in normal condition as it looks sharp as a knife.

Gadgetoid, 2017-11-24:
I'm not sure! Perhaps fbterm would help?

It would be tricky to adjust the scale- since anything that's not a 1:1 or 2:1 ratio would distort to the point of illegibility you're left with just: 800x480 or 400x240, although you might find the latter usable, I don't know how you would accomplish this on HyperPixel since it just accepts raw DPI signals from the Pi and can't do any scaling of its own.

You could, perhaps, experiment with setting:
```
framebuffer_width=400
framebuffer_height=240
```

snobu, 2017-11-24:
Looks like framebuffer settings just paint a smaller area of the display, there's no scaling. I gave up and installed minimal Xorg and xterm.

Sharing the xterm command line for "large enough font to see from 3m (that's 10ft across the pond)", should someone in the future work their way into a similar situation.

```bash
xterm -geometry 39x11 -rv -cr cyan -bd black \
          -fa 'Inconsolata' -fs 32 \
          -e '/app/start.sh'
```
[image]

Touchscreen just works in xterm which is very cool to highlight text just by touch it. I could probably make it scroll instead with a little effort..

Anyway @Gadgetoid thanks for looking at this issue! HyperPixel is fantastic, every time people see it on that Pi they ask if it's full HD :)

closed 2017-11-24 by snobu