pimoroni/hyperpixel #19: How to uninstall?

hoegge, 2017-11-30:
Reading the instructions I missed how to:

1) uninstall ?
2) switch back to HDMI without uninstalling?

I found the info here: 
[https://forums.pimoroni.com/t/hyperpixel-install-uninstall/5681/2](url)
but would be good to include on this site also to inform people up front.

Thanks

Xydis, 2017-12-28:
Broken link

Gadgetoid, 2018-02-15:
For posterity:

> You can see most of what makes HyperPixel tick in the file /boot/config.txt. Uncomment the lines which are marked as related to HyperPixel, and you’ll be back to HDMI.
> 
> You will also want to sudo systemctl disable hyperpixel-touch and sudo systemctl disable hyperpixel-init

Link should be: https://forums.pimoroni.com/t/hyperpixel-install-uninstall/5681/2

Gadgetoid, 2018-02-15:
I've also documented this in the README now: e043eb05b7fc6e3dc4c0e5293bc465c241f2e341

closed 2023-01-20 by Gadgetoid