pimoroni/hyperpixel #20: Android Things Support

apead, 2017-12-08:
Any plans for supporting Android Things?

Gadgetoid, 2017-12-08:
I'd love to, but unfortunately it's not an ecosystem with which I have any experience or expertise. The place to ask is probably here; https://github.com/androidthings/contrib-drivers/issues

apead, 2017-12-09:
Hi

Is there perhaps details available on how it works?

I could then possibly make it work with Android Things.  Other displays can
be set up on Things.

Regards
Allan

https://github.com/androidthings/contrib-drivers/issues

Gadgetoid, 2017-12-10:
It uses the DPI display output mode on the Raspberry Pi GPIO. If Android Things supports that, it should be straight-forward enough to get running I'd imagine.

The bigger problem is the touchscreen, which uses a software i2s bus on pins BCM10 and BCM11 (physical 19 and 23).

It involves putting the following in config.txt:

```
# For touch overlay:
dtoverlay=i2c-gpio,i2c_gpio_sda=10,i2c_gpio_scl=11,i2c_gpio_delay_us=4

# For LCD:
dtparam=i2c_arm=off
dtparam=spi=off

dtoverlay=hyperpixel
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=800
framebuffer_height=480
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87

dpi_output_format=0x6f016

display_rotate=2

hdmi_timings=800 0 50 20 50 480 1 3 2 3 0 0 0 60 0 32000000 6
```

And the dtoverlay switches the relevant pins into mode 6, DPI mode:

```
/dts-v1/;

/ {
	compatible = "brcm,bcm2708";

	fragment@0 {
		target = <0xdeadbeef>;

		__overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <0x1>;
		};
	};

	fragment@1 {
		target = <0xdeadbeef>;

		__overlay__ {

			dpi18_pins {
//				brcm,pins = <0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9    0xa 0xb 0xc 0xd 0xe 0xf 0x10 0x11 0x12 0x13       0x14 0x15>;
				brcm,pins = <0x0 0x1 0x2 0x3 0x4 0x5 0x6 0x7 0x8 0x9    0xc 0xd 0xe 0xf 0x10 0x11     0x14 0x15 0x16 0x17 0x18 0x19>;
				brcm,function = <0x6>;
				brcm,pull = <0x0>;
				phandle = <0x1>;
			};
		};
	};

	__symbols__ {
		dpi18_pins = "/fragment@1/__overlay__/dpi18_pins";
	};

	__local_fixups__ {

		fragment@0 {

			__overlay__ {
				pinctrl-0 = <0x0>;
			};
		};
	};

	__fixups__ {
		leds = "/fragment@0:target:0";
		gpio = "/fragment@1:target:0";
	};
};
```

A touchscreen driver would need to be written from scratch (using the Python as a reference) in either Java or perhaps C.

apead, 2017-12-10:
Hi

Awesome!!  Thanks!

Android Things 0.6 should support that.   The config.txt I have previously
populated with the settings on raspbian, however the problem was the
dtoverlay.

Touch screen would be a problem on Android Things, however could more
easily do it on a per application as a user driver in an Android service.
That's ok though as it's a single app OS by design and wouldn't really need
to worry anyways about anything else.

Will take a stab at doing it in Java.  And then of course do a xamarin c#
binding as that is what I do :).   I do the c# version of that Android
contrib library you shared.

If that works could look at doing a Windows 10 IoT version too.

Thanks again!

Regards
Allan

Gadgetoid, 2018-02-15:
@apead did you have any luck getting HyperPixel up and running with AndroidThings?

knobtviker, 2018-07-18:
Android Things does not support making kernel changes via config.txt, so adding drivers there is expected not to have any effect. (Copypasta of SO statement from [here](https://stackoverflow.com/questions/48541166/pcm5122-dac-with-android-things)).
I've written several drivers for Android Things and would like to tackle HyperPixel with touch support as it benefits me and my project at the most. However, I have no idea what the code below "And the dtoverlay switches the relevant pins into mode 6, DPI mode" does. 
Can it be done any other way, via Python code? Which I can then replicate into a driver.
A standalone driver for touch and HyperPixel is an acceptable path in Android Things ecosystem.

mrdavidli, 2019-03-05:
Hi, Anyone make progress writing a driver for Android Things for the HyperPixel?  If so, do you have a project link?  Xamarin C# would be neato.

apead, 2019-03-05:
I had a chat with a member of the Android Things team back in late 2017 and then stopped.

mrdavidli, 2019-03-05:
Thanx apead,  stopped because of something the Android Things said?  Wonder if anything changed two years later?

apead, 2019-03-05:
Yes to summarize it wasn't quite ready to do what I wanted to do.   I can have a look into again.  Rather a lot of time has passed.

frpalmisano, 2019-04-13:
Any idea how to proceed? I think the interest in this solution is strong. Let's make it work!

apead, 2019-06-27:
Google pulled the plug on Android Things.   So should we.

mrdavidli, 2019-06-27:
"Android Things continues to be a platform for experimenting with and building smart, connected devices using the Android Things SDK on top of popular hardware like the NXP i.MX7D and Raspberry Pi 3B. System images for these boards will remain available through the Android Things console where developers can create new builds and push app updates for up to 100 devices for non-commercial use."

Is that the most recent news?  February 19

apead, 2019-06-27:
There is this too:  "Note: Monthly security releases for the Raspberry Pi 3 and Pico i.MX7D will continue until August 2019."

Gadgetoid, 2019-11-05:
I think this is probably dead-in-the-water since Android Things appears to be a non-starter.

closed 2019-11-05 by Gadgetoid