pimoroni/hyperpixel #21: other OS support

Xydis, 2017-12-27:
Can you provide support to use this awesome display on Kali Linux on Pi?
Thank you!!

Gadgetoid, 2018-02-27:
I'm told it works on Kali, however you need to create a `/boot/config.txt` file and run the installer from this repository directly, like so:

```
sudo touch /boot/config.txt
git clone https://github.com/pimoroni/hyperpixel
cd hyperpixel
sudo ./setup.sh
```

Since this works I will update our one-line-installer to permit Kali and create a config.txt where appropriate.

closed 2018-02-27 by Gadgetoid

Massi361, 2021-03-16:
Hi, I tried to follow the directions, but it didn't work. To be precise, when I start the "setup.sh" it tells me that some packages do not exist and when I restart the Raspberry the screen stays black.
Could it be that with the passage of time this script no longer works?