pimoroni/hyperpixel #22: Multitouch support

bahamut657, 2017-12-27:
I see that the support in multitouch is partial (script hyperpixel-touch).
The software given read only 2 touch points (and sometimes is buggy).
Is it possible to read touch list indefinitely? (I know that this screen can read more than 2)
Regards
Jacopo

Gadgetoid, 2017-12-28:
I have some example code somewhere in the form of a custom driver that can read more than 2 touches, but the display is small enough that 2 is really the practical limit. Any more and it becomes tricky to distinguish between individual fingers, since - iirc - the sensor only has something like 14 vertical lines and 7 horizontal.

bahamut657, 2017-12-28:
Can u post some example code for enabling multitouch plz? 
Another thing with only 2 finger is:
1) start touch with finger A in position X,Y (near the top right side of the screen)
2) after a while start touching with finger B in position X-500, Y+200 (mantaining the finger A pressed)
=> Sometimes in the second touch arrive the X or the Y of the first finger

Gadgetoid, 2017-12-29:
Found it, at least this is how far I got: https://gist.github.com/Gadgetoid/64e57db4a22e4cd585ad35c6be2c7682

bahamut657, 2017-12-30:
I tried your code...but it seems to work better the original one.
With this new code, mantaining pressed a finger then pressing with the second one some point in the screen happens:
1) Sometimes touchend was triggered on the first finger (even if mantained pressed)
2) The second  finger hit has sometimes the wrong X or  Y
3) Is it possible to create a stable software managing at least 2 fingers?
I found an example in the Internet(managing indefinitely fingers and also simplifies the implementation)..Could we apply this to hyperpixel-touch script?
In case, I'm available to help you developing this piece of code if you want (this screen is wonderful but we need a stable driver for touchscreen)
The example code is : 
`#!/usr/bin/env python2
import time
import sys
import smbus

ADDR = 0x5c

bus = smbus.SMBus(3)

WINDOW_SIZE = 10

ROWS = 8
COLS = 14

H = 480
W = 800

STEP_Y = float(H) / (ROWS - 1)
STEP_X = float(W) / (COLS - 1)

data = [[0] * WINDOW_SIZE] * (ROWS + COLS)

def read_byte(address, delay=0):
    return bus.read_byte_data(0x5c,address)

def smbus_read_touch():
    raw_adc = bus.read_i2c_block_data(ADDR, 0x00, 32) + bus.read_i2c_block_data(ADDR, 0x20, 12)
    adc_out = [0] * (ROWS + COLS)

    y = 0
    for x in range(0,len(raw_adc),2):
        val = (raw_adc[x] << 8) | raw_adc[x+1]
        data[y].insert(0, val)
        data[y] = data[y][:WINDOW_SIZE]
        y += 1

    for x in range(ROWS+COLS):
        val = 0
        for y in range(WINDOW_SIZE):
            val += data[x][y]
        adc_out[x] = int(val / WINDOW_SIZE)

    touch_x = list(reversed(adc_out[8:]))
    touch_y = adc_out[:8]

    #print(str(int(time.time() * 1000)) + " y: " + " ".join([str(x).rjust(4,' ') for x in touch_y]) + " | x: " + " ".join([str(x).rjust(4,' ') for x in touch_x]))
    touches = []
    found = []

    ignore_factor = 0.6

    while True:
        max_x = max(touch_x)
        max_y = max(touch_y)

        io_x = touch_x.index(max_x)
        io_y = touch_y.index(max_y)

        if max_x < 100 or max_y < 100 or (io_x, io_y) in found:
            break

        io_x = touch_x.index(max_x)
        io_y = touch_y.index(max_y)

        found.append((io_x, io_y))

        b_x = (STEP_X * io_x) # + STEP_X / 2
        b_y = (STEP_Y * io_y) # + STEP_Y / 2

        if io_x < (COLS-1):
            b_x -= 1.0 - (float(touch_x[io_x + 1]) / touch_x[io_x]) * (STEP_X / 2)
            touch_x[io_x + 1] *= ignore_factor

        if io_x > 0:
            b_x += 1.0 - (float(touch_x[io_x - 1]) / touch_x[io_x]) * (STEP_X / 2)
            touch_x[io_x - 1] *= ignore_factor

        if io_y < (ROWS-1):
            b_y -= 1.0 - (float(touch_y[io_y + 1]) / touch_y[io_y]) * (STEP_Y / 2)
            touch_y[io_y + 1] *= ignore_factor

        if io_y > 0:
            b_y += 1.0 - (float(touch_y[io_y - 1]) / touch_y[io_y]) * (STEP_Y / 2)
            touch_y[io_y - 1] *= ignore_factor

        touch_x[io_x] *= ignore_factor
        touch_y[io_y] *= ignore_factor

        touches.append((int(b_x), int(b_y)))

    touches = sorted(touches, key=lambda touch: touch[0])
    print(touches)

def smbus_config_touch():
    bus.write_byte_data(0x5c,0x6e,0b00001110)

if __name__ == "__main__":
    smbus_config_touch()

    while True:
        smbus_read_touch()

        time.sleep(1.0/1600)
`
Regards,
Jacopo

Gadgetoid, 2018-01-03:
The attached code looks like another test I put together, where did you find it? I'm usually so buried under SD cards that these things go astray.

It had a problem, though; the relationship between reported touch ADC value, and distance to the actual row or col on the screen is not linear so as I moved my finger across the screen the position would accelerate and decelerate as it moved closer to/further away from each electrode. I need to find an algorithm that would map the nonlinear response of the ADC to a position between any two electrodes.

It also does not properly track touches, so a single touch at x=300,y=300 would be recognised as "Touch 1", but if you then touch at x=150,y=150 that would *replace* "Touch 1", and the original "Touch 1" would become "Touch 2."

bahamut657, 2018-01-03:
I think we can inspire to Android
[http://www.masonchang.com/blog/2014/8/25/androids-touch-resampling-algorithm](url)
Make me know if it is ueful and if I can help you.
Regards,
Jacopo

Gadgetoid, 2018-01-04:
I've put together a basic pygame app for getting a better idea of the touch data and recognition. You can find it here: https://gist.github.com/Gadgetoid/8bc3bbbd64d7f96f585920eb1ef6abde

Unfortunately it's shown just how many bugs are in my current detection, it's pretty poor. It also makes no effort at all to actually track fingers across the screen, it just recognises raw points of touch which results in all sorts of problems.

bahamut657, 2018-01-05:
Thanks, I will try it soon

closed 2023-01-20 by Gadgetoid