pimoroni/hyperpixel #28: Prevent HyperPixel from going to sleep

snobu, 2018-03-05:
I've tried everything but seems that after 10-15 minutes no matter what i do with xset or what parameters i pass to Xorg, the HyperPixel goes to sleep and only wakes up if i manually call:

```
xset s activate
```

What i've tried:

```
Xorg -dpms -nocursor 2>&1 > /dev/null &
xset s noblank
xset s 0
xset s off
xset -dpms
```

Is there a flag i can set in `config.txt` to prevent power saving from happening? Is it expected behavior?

I really mean power saving not blanking because the LCD back light goes off as well.

Other than this, it's probably the best thing ever built for the Pi. Period.

snobu, 2018-03-05:
To clarify, it goes to sleep only with no keyboard activity not randomly.

Gadgetoid, 2018-03-05:
I'm not entirely sure how to accomplish this, there's an alternate solution documented here though:

https://www.raspberrypi.org/forums/viewtopic.php?t=158976

Add:

```
Section "ServerFlags"
Option "blank time" "0"
Option "standby time" "0"
Option "suspend time" "0"
Option "off time" "0"
EndSection 
```

Into `/etc/X11/xorg.conf`

Which might work!

snobu, 2018-03-05:
Hmm I naively assumed xset does exactly that but it may not touch the scope of that  Xorg that I manually launch. Let me give it a shot. Thanks!

snobu, 2018-03-06:
Those `ServerFlags` seem to have no effect on my Xorg (i did check and it was picking up the right configuration file). May be the right solution but just didn't work for me.

I fixed it by adding this to `/etc/rc.local`, in addition to the `xset`s in my original post.

```
xset dpms force on
```

From `man xset`:

> The `force` flags forces the server to immediately switch to the DPMS state specified.  The DPMS state can be one of `standby', `suspend', `off', or `on'.

@Gadgetoid If you're interested in adding that to the HyperPixel README.md i can send a PR.

closed 2018-03-06 by snobu

Gadgetoid, 2018-03-06:
@snobu glad you found a solution in the end! I, and I'm sure future users of HyperPixel, would appreciate a PR. Thank you!