pimoroni/hyperpixel #32: Enable easy switch between hyperpixel and HDMI

hoegge, 2018-03-13:
It would be very useful if there was an easy way to disable the display (i.e. enable HDMI again) and enable (when done using HDMI) like a script so you do not have to uninstall every time you need to change display. Would that be easy to do?

Thanks

Gadgetoid, 2018-03-13:
There's a good topic about this on our forums, here: https://forums.pimoroni.com/t/hyperpixel-selective-output-eg-on-boot/5193

hoegge, 2018-06-12:
Is there any way to use the HDMI as main display and then only use the HyperPixel for dedicated programs? The hyperpixel display is not very suitable for debugging and developing.

tinspin, 2020-01-23:
If you write a 1 to this file /sys/class/backlight/rpi_backlight/bl_power the screen turns off... write a 0 again to turn it on... and this works in realtime!

closed 2023-01-20 by Gadgetoid