pimoroni/hyperpixel #33: Multitouch Kernel driver

procount, 2018-03-15:
I decided to __*try*__ and create a C based kernel driver for the Hyperpixel touchscreen interface.
It's quite tricky to detect the touches, isn't it?
1. How to avoid ghosting when reading the ADC values directly? I see the hyperpixel_viewer exhibits this behaviour. Now I understand why you use the 0x40 registers to get the detected co-ords. I think the ADC values are just used for touch confirmation in this driver, I think?
2. The 2 detected touchpoints seem to be reported in order of their distance from the origin(?), rather than order of touching, so some additional tracking ID support is required.
3. Because of (2), I think the button up/down events can get confused.
4. Evtest seems to lose (or not report) some event messages if they are not required (e.g. some MT_SLOT events seem to go missing, even though the driver outputs them, redundant ones are removed?)

To get >2 touches, it seems necessary to use the raw ADC values, but ghosting seems to be a real issue.
Unless you have any more knowledge on how to avoid this, I will probably go down the route of just porting this python driver for 2 touches, but try to improve the tracking of the touches.

Gadgetoid, 2018-03-15:
I believe using the raw ADC values is the key to making a good driver, but you're right about ghosting. The trouble is that's the challenge- coming up with an algorithm, or touch recognition technique that takes it into account and compensates for it. The built-in two-touch recognition works, just about, but has some key weaknesses when recognising two touches that make gestures a little clumsy and make certain things - like the Pong demo and analog-stick spaceship demo I made for the official touchscreen - impossible.

As a tangent- the raw ADC values are extremely useful for triggering precisely placed controls. I managed to create a rough button driver that makes parts of the screen behave like a D-Pad and A/B buttons.

But I digress- I don't have any deep insight into how we'd solve the ghosting issue and recognise/track 2-4 touches with any degree of accuracy.

procount, 2018-03-15:
I've not see your Pong demo or analog-stick spaceship demo - do you have a link. Maybe I could use them as a target/goal ?

Gadgetoid, 2018-03-15:
They are part of python-multitouch, which is for the official Raspberry Pi 7" touchscreen- https://github.com/pimoroni/python-multitouch

Since HyperPixel is the same resolution, they should look fine but - if memory serves me - do not work particularly well at all.

procount, 2018-04-29:
I've added support for the Hyperpixel display to PINN.
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=142574&p=1309584#p1309584
No touchscreen support yet, but I hope to add that later when I find a bit more time.

closed 2018-06-12 by Gadgetoid

procount, 2019-02-01:
@Gadgetoid - OK, I'm back on creating a kernel touchscreen driver for the Hyperpixel 3.2.
I'm making good progress now that my driver loads, detects the touchscreen is fitted and can read the touch positions.

Do you have any more information on the following registers:

0x6e: interrupt settings - 0 0 0 0 EN_INT INT_POL INT_MODE1 INT_MODE2
0x6f?: interrupt width
0x70: power mode 0000 <- IDLE PERIOD, 1 ALLOW SLEEP 00 <- POWER MODE

particularly the interrupt modes.

I see your python driver uses:
  EN_INT = 1 (Enable Interrupt?)
  INT_POL = 1 (Interrupt Polling?)
  INT_MODE1 = 1 (Interrupt mode 1?)
  INT_MODE2 = 0 (Interrupt mode 2?)
and polls the interrupt line. What clears the interrupt line? Is it when you read from registers 0x40 onwards?
Do you know how to set this up for proper interrupt mode? 
What are the other interrupt modes it supports?
I guess you have no idea about 0x6f (interrupt width) as you already put a question mark against it.

I don't see the HP 3.2 on your shop anymore. Is it discontinued now the HP4 is available?