pimoroni/hyperpixel #34: Display switches to lowest brightness on boot with no way to change it back

SamusAranX, 2018-04-05:
I just set up my HyperPixel, but it's currently in an unusable state.

While booting, the brightness is at an acceptable level, but once the console output reaches the login mask, the display switches to max brightness for about 2 frames, then switches back to minimum brightness and stays there. I tried messing with `/sys/class/backlight/rpi_backlight/max_brightness`, but it seems the only acceptable values for that are zero for off and one for on.

Is there anything I can do to fix the backlight?

I'm using the HyperPixel with a Raspberry Pi 2B with a fairly recent install of Raspbian Lite.

Gadgetoid, 2018-04-06:
Are you running any code that could be interfering with the backlight control pin on BCM19? The backlight is set to on by default with a pull resistor.

By default the backlight can only be toggled on or off, since the generic PWM backlight control module required to make it dimmable isn't available in Raspbian. (I'm still chipping away at that)

SamusAranX, 2018-04-06:
This is a fresh install of Raspbian Stretch. Unless there's something in there that actively interferes with the HyperPixel, no.

Additionally, the display won't display anything now. I reinstalled Raspbian from scratch and ran this repo's `setup.sh`, but even after multiple reboots, the display's staying black. Ironically, the brightness is at the desired level now.

I found [this post of yours](https://forums.pimoroni.com/t/hyperpixel-nothing-on-screen-after-reboot-for-pi3-fresh-raspbian/5597/2) and tried running `hyperpixel-init`, but that does nothing. `dmesg`'s output doesn't contain anything interesting, either.

Any ideas? 😕

Gadgetoid, 2018-04-09:
There's nothing that interferes in the latest Stretch, as far as I'm aware. The hyperpixel-init script should be starting on boot (you could try: `sudo systemctl enable hyperpixel-init` just to be sure).

Gadgetoid, 2020-03-10:
Closing this issue for now since it's gone stale, but please let me know if you found a solution or are still having trouble.

closed 2020-03-10 by Gadgetoid