pimoroni/hyperpixel #37: How to configure on dual boot

ZoltXs, 2018-05-14:
Hello I would like to know if I can configure hyperpixel in dual boot?

Gadgetoid, 2018-05-15:
You're going to have to be waaaay more specific than that- how are you dual booting, and what operating systems? Bear in mind that HyperPixel is only fully supported in Raspbian and OSes that are built upon it.

ZoltXs, 2018-05-16:
Using noobs, I install Raspbian, Recalbox, when I do reboot I can not see the main menu to choose Boot option.

Gadgetoid, 2018-05-18:
noobs has its own config.txt if I remember correctly, you would need to add the hyperpixel settings into this file too. The boot menu would only show upon reboot, though, since upon a cold boot hyperpixel requires an initialisation sequence to be run by the OS.

For getting the menu on cold boot you might find PINN (although before now I've not heard of it) works better than NOOBS: https://github.com/procount/pinn  - they added a method to run third party init scripts:

https://github.com/procount/pinn/commit/d3124a9c7385ebf8e9ff1bdeb531ba6b1355a4c9

hyperpixel-init is a statically-linked binary init routine that sends the correct init sequence and should be possible to run in PINN: https://github.com/pimoroni/hyperpixel/blob/master/requirements/usr/bin/hyperpixel-init

Note: I've not tried any of this!