pimoroni/hyperpixel #47: Does not work on current Buster Pi4 1GB

HansOtten, 2020-04-09:
To check out the Pi4 and current Buster (feb 2020) I did the followoing:

0 Checked on a Raspberry Pi 3, current Buster, instructiosn followed and works fine, so hardware is OK

1 imaged  a SD with current Buster (feb 2020)
2 booted on a Pi 4 1 GB
3 did the usual raspi-config things and update/upgrade
4  checked out the system, works as expected
5 curl https://get.pimoroni.com/hyperpixel | bash
6. rebooted , installed Hyperpixel 

Result is rubbish, reverse image on HDMI, Hyperpixel dead

Maybe the curl https://get.pimoroni.com/hyperpixel | bash gets the old hyperpixel install and the the Pi 4? Looks like it.

So back to 0,

1 imaged  a SD with current Buster (feb 2020)
2 booted on a Pi 4 1 GB
3 did the usual raspi-config things and update/upgrade
4  checked out the system, works as expected
5 downloaded the Hyperpixel Pi 4 git repository
6. rebooted , installed Hyperpixel with ./setup.h

Result is Hyperpixel noise banner, HDMI shows normal screen

So install fails with the curl or setup ,and  instructions clearly wrong.

Gadgetoid, 2020-04-14:
Are you using an old original HyperPixel or a HyperPixel 4?

HansOtten, 2020-04-14:
Hi Phil!

 

Hyperpixel 3.5, the original one! Old but a good picture!

 

Hans

tonywaite, 2020-04-16:
Hi,

Same for me using a Pi4 and Hyperpixel 4.0.
(The Hyperpixel 4.0 hardware works perfectly when imaged for the Pi3B)

T o n y

Gadgetoid, 2020-04-16:
@HansOtten - I will have to try and find a HyperPixel 3.5 to try and replicate this- working from home doesn't make that easy. My apologies if I can't get back to you imminently.

@tonywaite - for HyperPixel 4.0 you should go here: https://github.com/pimoroni/hyperpixel4

HansOtten, 2020-04-17:
No need for hurry or apologies, working from home and staying safe is more important. It can wait!

Gadgetoid, 2020-10-19:
Just realised I haven't got back to this. Sorry! @HansOtten have you had any luck since?

HansOtten, 2020-10-19:
Hi Phil,

 

I have waited with patience, realizing how tough life is traveling to the office is since the pandemic!

 

So I am still interested in and thankful for a solution.

Hans

Gadgetoid, 2020-10-19:
I noticed with step 5 you said:

5. downloaded the Hyperpixel Pi 4 git repository

Did you download *this* GitHub repository, or the HyperPixel4 one? Since 4 is for the 4" HyperPixel and not the Pi 4.

That said, the curl installer *should* have worked. It's possible that the hyperpixel init routine did not work, or that something else failed. I will see if I can find my old HyperPixel now.

HansOtten, 2020-10-19:
I did use the  Hyperpixel git, not the Hyperpixel 4.

The problem is with the old Hyperpixel on Buster as is indicated in the git. Runs fine on older OS than Buster. 

 

Hans

Gadgetoid, 2020-10-19:
I've managed to get it working on Buster here using the pi4 branch of this repository, but I'm seeing exactly the same problem you describe as "noise banner."

Running `hyperpixel-init` manually once the system has booted seems to fix this, but that's not ideal.

I've no idea why the `hyperpixel-init` script isn't setting up the LCD correctly in the first place. Currently experimenting!

calini, 2020-10-23:
@Gadgetoid am currently experiencing the same issue with a Pi 4 running Buster and a HyperPixel 3.5". After `hyperpixel-init` the screen seems to work fine.

Gadgetoid, 2020-10-24:
@calini have you tried grabbing and installing from GitHub, the "pi4" branch? I had to make this change to get touch working on my Pi 4, and I'm not sure if the one-line-installer uses the Pi4 branch at all - https://github.com/pimoroni/hyperpixel/commit/347e07f6a03e20bc89361b6e92bd28e19c9bfe47

calini, 2020-10-24:
@Gadgetoid I should have probably mentioned that too: I got the one line installer and edited it to pull the `pi4` branch instead of `master`

veproza, 2021-02-14:
For those stumbling here from Google, a workaround is mentioned in [parallel issue](https://github.com/pimoroni/hyperpixel/issues/50#issuecomment-703083644) - use an older kernel

`sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

HansOtten, 2021-02-14:
Any progress? Or is the Hyperpixel 3.5 ripe for the museum?

hmax42, 2021-05-26:
seems not :(
and i hoped to use it with my pi 400

HughMungis, 2023-01-07:
created a new issue but scrolled far enough down to see this. hopefully it's a useful bump back to the top

closed 2023-01-20 by Gadgetoid