pimoroni/hyperpixel #48: Shutdown button

sgrunt91, 2020-05-04:
Hello,
I've successfully installed the Hyperpixel 4 screen, thanks for you work !

But I've a little question, is it possible to use a GPIO pin to wire button that start a safe shutdown script ? or all the pins are used by the screen ? 
If not, how do you do to safe shutdown the Pi with a button and hyperpixel ?

Regards,
Sgrunt

Gadgetoid, 2020-07-31:
Every single pin is used, I'm afraid, though you might have some luck grabbing BCM 26 shortly after boot (you'll have to delay for a few seconds, or make sure whatever starts up your script checks that the `hyperpixel4-init` systemd unit has run first), pulling it high with the internal pull-up and connecting to ground via a push button.

BCM 26 is "LCD Program Data" which is the data pin of the SPI connection used to set up the LCD. Once the LCD is set up, however, it's "unused". That's so say, the Pi doesn't actively drive it, but it's still connected to the LCD.

closed 2020-07-31 by Gadgetoid

baesek, 2021-11-24:
> Hello, I've successfully installed the Hyperpixel 4 screen, thanks a lot for your work !
> 
> But I've a little question, is it possible to use a GPIO pin to wire button that start a safe shutdown script ? or all the pins are used by the screen ? If not, how do you do to safe shutdown the Pi with a button and hyperpixel ?
> 
> Regards, Sgrunt

Hey would you share the script you came up with to use a Shutdown Button with the hyperpixel on GPIO 26? I am having the exact same problem right now.