pimoroni/hyperpixel #50: Touch not working on Pi 3B with Buster

gazzmanzx6, 2020-09-26:
Having just updated my Pi 3B to the latest Buster build the touch screen has stopped working. It was fine before.
I have tried reinstalling the drivers but touch still doesn't work.
Is there anything I can do to debug this?

gazzmanzx6, 2020-10-03:
It appears this is to do with the latest kernel. It also affects the Pi 2B as well.
Rolling back to 4.19.118 makes the touch work again.

Gadgetoid, 2020-10-25:
Might be worth trying the Pi 4 branch of this GitHub repository. I'm not sure if the fixes I had to make there translate across but it's highly likely.

One gotcha between major kernel releases is that the pre-compiled `dtbo` files might no longer be compatible. I've changed the "pi4" branch here to compile them from source, instead.

gazzmanzx6, 2020-10-27:
Unfortunately that doesn't work. Just some garbled pixels on the screen.

Gadgetoid, 2020-10-27:
> Didn't work at first but running hyperpixel-init seems to have fixed it.

ugh I need to figure out why the at-boot `hyperpixel-init` isn't working. But glad this got you up and running most of the way.

closed 2023-01-20 by Gadgetoid