pimoroni/hyperpixel #52: How to install hyperpixel on Kali linux?

Massi361, 2021-03-17:
I would like to understand if this display is compatible with Kali linux.
I have tried everything I have found on the internet, but none have worked.

I tried to use the command "curl https://get.pimoroni.com/hyperpixel | bash", but it tells me that the hardware is not compatible.
I tried to use the indications present in a previous "problem", but in the terminal I am told that some packages are unknown and if I restart Rasperry, the screen remains black and kali no longer works.
I tried to do the manual installation, respecting each step (what is written in the "code" section), but the screen remains black.

hzykiss, 2022-09-07:
I have the same problem as you. Have you solved it?

Massi361, 2022-10-15:
I finally managed to solve the problem.
hzykiss, if it helps, I'll explain how I did it:
- Use the command "curl -sSL get.pimoroni.com/hyperpixel4-legacy | bash" to do the automatic installation and when it asks you to reboot the raspberry, don't do it.
- Make sure you have installed "RPi.GPIO" (to check, type "python3" in the terminal and then "import RPi.GPIO as GPIO"; if it doesn't give you an error, it means it's installed) otherwise in the terminal type "pip install RPi.GPIO".
- In the terminal, type "sudo apt install python3-dev python3-rpi.gpio" to install the following packages, if you don't already have them installed.
- Enter the file "hyperpixel4-init" by typing "sudo nano /usr/bin/hyperpixel4-init" in the terminal and in the first line ("#!/usr/bin/env python") add a "3" attached to the word "python" so that the script uses Python3.
- Restart the raspberry and the screen should turn on, if this does not happen type in the terminal "systemctl status hyperpixel4-init.service" and see what errors it gives.
I hope I have been of some help.

closed 2023-01-20 by Gadgetoid

dkyazzentwatwa, 2024-05-31:
> I finally managed to solve the problem. hzykiss, if it helps, I'll explain how I did it:
>
> * Use the command "curl -sSL get.pimoroni.com/hyperpixel4-legacy | bash" to do the automatic installation [...]

I was having so much trouble following the official instructions and followed yours step by step...closed my eyes and pressed sudo reboot...prayed for the kali dragon. And there it is!

Thank you Messi!