pimoroni/hyperpixel #54: bullseye convert hyperpixel-touch script to python3

pwr33, 2021-11-13:
python2-pip gone and that sort of problem, so convert it to python3?

simply change the shebang to python3

basically you use some odd old python2 function file(), converting to python open() works ok except for the se file which fails on the os.dup().  cant find the old python documentation on the file() function it is that old...

any ideas

Gadgetoid, 2021-11-16:
Looks like this is a very similar set of changes - https://github.com/pimoroni/phat-beat/pull/37/files

For stderr:

```python3
se = open("/dev/null", "ab+", 0)
os.dup2(se.fileno(), sys.stderr.fileno())
```

Works for me on Python 3.8

pwr33, 2022-02-04:
Hi, been a while, just spent a bit more time on it, it is now failing on the line below
bus = smbus.SMBus(3)

does it need a udev rule or something?
there is no /dev/i2c-3

====

  File "/usr/bin/hyperpixel-touch", line 117, in <module>
    bus = smbus.SMBus(3)
FileNotFoundError: [Errno 2] No such file or directory
Exception ignored in: <function InputDevice.__del__ at 0xb612ebb0>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3/dist-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3/dist-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.9/asyncio/events.py", line 725, in get_event_loop_policy
  File "/usr/lib/python3.9/asyncio/events.py", line 718, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down

pwr33, 2022-10-05:
OK, it is now SMBus(11) on buster and bullseye

And to stop a segfault, change

/usr/lib/systemd/system/triggerhappy.socket
[Socket]
remove ListenStream=/run/thd.socket
replace with ListenDatagram=/run/thd.socket

works for me anyway