pimoroni/hyperpixel #57: Hyperpixel (original one) on rpi 2 v1.2 bullseye

hmax42, 2022-03-31:
Hi

I dug out my RPi 2 v1.2 with the Hyperpixel, it had a Jessie installation, armhf as there was no arm64 at that time

i updated it to bullseye-arm64 (!) (new sd card) and checked out this repo
did not work, dependecies are missing
so i switched to the rpi4 branch
same errors, lots of armhf complaints

> pi@raspberrypi:~/hyperpixel $ sudo ./setup.sh
> Installing dependencies...
> (Reading database ... 37817 files and directories currently installed.)
> Preparing to unpack .../python-evdev_0.6.4-1_armhf.deb ...
> Unpacking python-evdev:armhf (0.6.4-1) over (0.6.4-1) ...
> dpkg: dependency problems prevent configuration of python-evdev:armhf:
>  python-evdev:armhf depends on libc6 (>= 2.4).
>  python-evdev:armhf depends on python (>= 2.7).
>  python-evdev:armhf depends on python (<< 2.8).
> 
> dpkg: error processing package python-evdev:armhf (--install):
>  dependency problems - leaving unconfigured
> Errors were encountered while processing:
>  python-evdev:armhf
> python-rpi.gpio is required
> Updating apt indexes...
> .....................Installing python-rpi.gpio...
> E: Package 'python-rpi.gpio' has no installation candidate
> Apt failed to install python-rpi.gpio!
> Falling back on pypi...
> python-smbus is required
> Installing python-smbus...
> E: Package 'python-smbus' has no installation candidate
> Apt failed to install python-smbus!
> Falling back on pypi...
> 
> Installing Requirements...
> 
> Building & installing hyperpixel.dtbo
> 
> Building & installing hyperpixel-gpio-backlight.dtbo
> 
> Installing init script...
> 
> All done!
> 
> Some changes made to your system require
> your computer to reboot to take effect.
> 
> Would you like to reboot now? [y/N] y

but wonder, it showed the starting text console and some raspberries....

...for a few seconds, then everything dimmed away, while showing some vertical lines. then they are gone too
i would not call it noise as other issues have called it

i attached a keyboard to check if it is just a screensaver, but it is not.

running hyperpixel-init manuelle just flashes the scren for a moment, but the text does not return.

hoegge, 2022-08-24:
HyperPixel is not supported on Bullseye and forward. You must upgrade to a HP 4.0 for that.

hmax42, 2022-08-24:
not liking this, but i know it's the sad truth.

closed 2022-08-24 by hmax42