pimoroni/hyperpixel #60: HyperPixel OS Compatibility (NOT compatible with Bullseye/device-tree)

Gadgetoid, 2023-01-20:
HyperPixel was developed for a very different Raspbian than the Raspberry Pi OS which now powers Pi's.

Screens have shifted from being a Pi-specific hack in `config.txt` which causes the VideoCore chip to barf out DPI signals which we could shore up with Python scripts and other such (very very wrong in retrospect) nonsense, to a much more stringent Linux device-tree configuration.

As such we can't support HyperPixel on Bullseye. The conflicting pin assignments make it - if not impossible - extremely contrived.

Support on Buster is possible via a Python tweak found here: https://github.com/pimoroni/hyperpixel/issues/54

tomgidden, 2023-01-25:
Is this just in relation to touch input, or the whole thing?

I got sporadic display on my non-touch HyperPixel (original) on the latest Bullseye Lite OS (as of today) but after the boot-up scroll, it goes blank with one of those disturbing oh-no-my-display-is-broken garbage fades.

I'm reflashing the legacy OS now in the hope that I haven't borked the screen somehow. HyperPixel 4 won't fit in the enclosure I use.

HughMungis, 2023-02-18:
Is there a pre-bullseye build that you all recommend as a "most recent working" OS? I don't mind ditching bullseye.

theschles, 2023-08-25:
@Gadgetoid  I've incorporated your comments above into PR https://github.com/pimoroni/hyperpixel/pull/61 

Please please please consider implementing the "contrived pin assignments" -- or whatever it takes -- for the HyperPixel 2.1 so it will be 100% supported without additional hacking around in Bullseye-based and Bookwarm-based Raspbian OS !  

This is a great display.  I deliberately bought the display [after watching @MattGrayYes 's video using the HyperPixel 2.1 to display pictures of the "Blue Marble" (a.k.a. "Earth") from the EPIC library]{https://www.youtube.com/watch?v=q4M3cm4ml_c}.  My dad's 82nd birthday is on Sunday and I'm making the project [using MattGreyYes's code]{https://github.com/MattGrayYes/epic} as a birthday gift.

Or please design / implement / release a _new_ round HyperPixel which does support Bullseye/Bookworm-based Raspbian OS before Buster goes End of Life on June 30, 2024.  I'd gladly buy that and swap out this HyperPixel 2.1.